tarm/serial #20: Builds serial_posix instead of serial_linux on Linux amd64

Simon-L, 2015-02-27:
Hi,
Seems this package builds for posix on linux.
Here is the output of the build even with `GOOS=linux`, as you can see it builds the posix file.
Could it be an issue with the `// +build` build constraints?
For now I've tweaked the linux and posix files so that the linux one builds with "cgo". (removed the excluding `!` )
```
$ GOOS=linux go install -x github.com/tarm/serial
WORK=/tmp/go-build450422516
mkdir -p $WORK/github.com/tarm/serial/_obj/
mkdir -p $WORK/github.com/tarm/
cd /home/simon/go/src/github.com/tarm/serial
CGO_LDFLAGS="-g" "-O2" /usr/local/go/pkg/tool/linux_amd64/cgo -objdir $WORK/github.com/tarm/serial/_obj/ -- -I $WORK/github.com/tarm/serial/_obj/ serial_posix.go
/usr/local/go/pkg/tool/linux_amd64/6c -F -V -w -trimpath $WORK -I $WORK/github.com/tarm/serial/_obj/ -I /usr/local/go/pkg/linux_amd64 -o $WORK/github.com/tarm/serial/_obj/_cgo_defun.6 -D GOOS_linux -D GOARCH_amd64 $WORK/github.com/tarm/serial/_obj/_cgo_defun.c
gcc -I . -fPIC -m64 -pthread -fmessage-length=0 -print-libgcc-file-name
gcc -I . -fPIC -m64 -pthread -fmessage-length=0 -I $WORK/github.com/tarm/serial/_obj/ -g -O2 -o $WORK/github.com/tarm/serial/_obj/_cgo_main.o -c $WORK/github.com/tarm/serial/_obj/_cgo_main.c
gcc -I . -fPIC -m64 -pthread -fmessage-length=0 -I $WORK/github.com/tarm/serial/_obj/ -g -O2 -o $WORK/github.com/tarm/serial/_obj/_cgo_export.o -c $WORK/github.com/tarm/serial/_obj/_cgo_export.c
gcc -I . -fPIC -m64 -pthread -fmessage-length=0 -I $WORK/github.com/tarm/serial/_obj/ -g -O2 -o $WORK/github.com/tarm/serial/_obj/serial_posix.cgo2.o -c $WORK/github.com/tarm/serial/_obj/serial_posix.cgo2.c
gcc -I . -fPIC -m64 -pthread -fmessage-length=0 -o $WORK/github.com/tarm/serial/_obj/_cgo_.o $WORK/github.com/tarm/serial/_obj/_cgo_main.o $WORK/github.com/tarm/serial/_obj/_cgo_export.o $WORK/github.com/tarm/serial/_obj/serial_posix.cgo2.o -g -O2
/usr/local/go/pkg/tool/linux_amd64/cgo -objdir $WORK/github.com/tarm/serial/_obj/ -dynimport $WORK/github.com/tarm/serial/_obj/_cgo_.o -dynout $WORK/github.com/tarm/serial/_obj/_cgo_import.c
/usr/local/go/pkg/tool/linux_amd64/6c -F -V -w -trimpath $WORK -I $WORK/github.com/tarm/serial/_obj/ -I /usr/local/go/pkg/linux_amd64 -o $WORK/github.com/tarm/serial/_obj/_cgo_import.6 -D GOOS_linux -D GOARCH_amd64 $WORK/github.com/tarm/serial/_obj/_cgo_import.c
gcc -I . -fPIC -m64 -pthread -fmessage-length=0 -o $WORK/github.com/tarm/serial/_obj/_all.o $WORK/github.com/tarm/serial/_obj/_cgo_export.o $WORK/github.com/tarm/serial/_obj/serial_posix.cgo2.o -g -O2 -Wl,-r -nostdlib /usr/lib/gcc/x86_64-linux-gnu/4.8/libgcc.a -Wl,--build-id=none
/usr/local/go/pkg/tool/linux_amd64/6g -o $WORK/github.com/tarm/serial.a -trimpath $WORK -p github.com/tarm/serial -D _/home/simon/go/src/github.com/tarm/serial -I $WORK -pack ./serial.go $WORK/github.com/tarm/serial/_obj/_cgo_gotypes.go $WORK/github.com/tarm/serial/_obj/serial_posix.cgo1.go
pack r $WORK/github.com/tarm/serial.a $WORK/github.com/tarm/serial/_obj/_cgo_import.6 $WORK/github.com/tarm/serial/_obj/_cgo_defun.6 $WORK/github.com/tarm/serial/_obj/_all.o # internal
mkdir -p /home/simon/go/pkg/linux_amd64/github.com/tarm/
mv $WORK/github.com/tarm/serial.a /home/simon/go/pkg/linux_amd64/github.com/tarm/serial.a
```

cormacc, 2015-12-10:
I'm pretty sure serial_linux is deprecated, though perhaps tarm can confirm. Noticed the same thing a while back and spent some time scratching my head and messing with it before realising it looks like an earlier WIP version of serial_posix.

tarm, 2015-12-10:
Sorry for not replying to this issue sooner.

No, not deprecated.  Depending on your constraints, it may be easier to build with syscalls or with cgo.

The _posix version uses cgo.  That can be great and should work across linux, OSX, bsd, etc, but only if you are doing a native compile.  If you are trying to cross-compile, it will probably be much more difficult.  

The _linux version should allow you to cross compile your program to linux (likely just x64 + x86 right now, but that needs to be fixed.  See #35).  That can be handy if you want to develop on OSX but cross-compile for linux.  You can also cross-compile to windows as well since the windows version does not use cgo.

A linux x86 build should function the same with either _posix or _linux.  The default was chosen to be _linux since that avoids cgo and is perhaps slightly cleaner.  My inclination is to keep the cgo/syscall split the way it is, but I'm open to suggestions about changing the default to cgo.

trusch, 2016-06-20:
The initial problem that the posix version is prefered on linux hosts is still in place. This prevents me from using baudrates > 115200 Baud. 
Can this be fixed? Otherwise I would fork this lib an tweak the //build instructions mentioned by Simon-L.