tarm/serial #24: Serial Port Configurability

ngtrimble, 2015-06-03:
I'm curious, from a code point of view what it would take to add configuration for data bits, parity and stop bits. Is there some hurdle to this being implemented? Asking before I start digging in for possible solutions. Sounds like you are open to contributions?

tarm, 2015-06-04:
I am certainly open to pull requests.  I am not actively working on adding features, so most recent contributions have come from other people.

Configuration with more options should be relatively straightforward.  I would like to see:
- Identical feature support on all platforms: linux, posix, osx, windows
- Testing on several platforms including at least linux and windows
- Backwards compatibility: if unset, the default should be for 8 bits, etc.
- Implementation in the Config struct.  Some commented out starting points are here: https://github.com/tarm/serial/blob/master/serial.go#L77
- Clean patches, which may mean rebasing your commits for bug fixes and after code reviews.

Several people have made forks and made changes to add those features but have never submitted pull requests.  I would start with those changes and either use them for inspiration or clean-up the patches and use them directly.  https://github.com/tarm/serial/network

tarm, 2016-01-31:
@tajtiattila made a PR that adds this functionality and I just merged it to the master branch.

Please test and let us know if you have any problems.

closed 2016-01-31 by tarm