tarm/serial #26: Can't build for OS X on Travis

saulshanabrook, 2015-06-30:
First off, thanks for the awesome library :)

I seem to be having a problem building my go app (which imports this library) on travis for darwin/amd64: https://travis-ci.org/lucibus/subicul/builds/69061820#L213-L214

However, when I try to build for that locally on my mac it works fine. Is it possible to build this for darwin when one is not on a mac?

tarm, 2015-07-01:
I looks like you are cross compiling from linux to OSX?

I'm not surprised that does not work.  The OSX implementation uses cgo.

The windows implementation does not use cgo.  Also there is a linux implementation that uses syscalls instead of cgo (see serial_linux.go in the directory and look at the build tags at the top of the file).

You could probably make an OSX implementation that does not use CGO and send a PR.

Thanks,
Tarm

saulshanabrook, 2015-07-01:
Thanks for the response. Yeah this is my first time compiling go, so I realized that it was the cgo tripping it up. Sounds good.

closed 2015-07-01 by saulshanabrook