tarm/serial #27: Read function always EOF

a-giovani17, 2015-10-06:
Hello, I tried exactly as the example (except the port name). I use tty0tty as the serial port. I successfully write string to the port but failed to read from it. It always return an empty string and raise the "EOF" error. I've used time interval to wait for the writing process but no luck. 

Any suggestions?

tarm, 2015-10-07:
What platform are you running on?  Can you share your code or an example that illustrates the issue?

By default Reads should be blocking.

a-giovani17, 2015-10-08:
I use Ubuntu 14.04 64-bit OS. Here is my code

```go
package main

import (
        "log" 
        "serial"
        "time" 
        "fmt"
)

func main() {
        c := &serial.Config{Name: "/dev/tty0", Baud: 115200, ReadTimeout: time.Second * 5}
        s, err := serial.OpenPort(c)
        if err != nil {
                log.Fatal(err)
        }

        n, err := s.Write([]byte("test"))
        fmt.Println("start writing..")
        if err != nil {
                log.Fatal(err)
        }
        
        time.Sleep(time.Second/2)
        fmt.Println("finished writing..")

        buf := make([]byte, 128)
        n, err = s.Read(buf)
        if err != nil {
                log.Fatal(err)
        }
        log.Printf("%q", buf[:n])
}
```

tarm, 2015-10-09:
Do you have something connected that is supposed to be writing to the other side of the serial port?

I assume you are seeing the log.Fatal() after the Read after 5 seconds?

What happens if you do not set the ReadTimeout in the config?

The way the serial port works, if the read times out it will probably return an EOF instead of a timeout error.

kgrvamsi, 2020-03-06:
i have a question here on how can you determine the read took all the output that it needs from the command you send?

xiegeo, 2020-03-09:
@kgrvamsi There is no general solution. You need to read the menu on the protocol details your device is using.