tarm/serial #35: Installation fails on Raspberry Pi

fixermark, 2015-12-07:
sudo go get github.com/tarm/serial
# github.com/tarm/serial
/usr/lib/go/src/pkg/github.com/tarm/serial/serial_linux.go:114: undefined: syscall.TCIOFLUSH

tarm, 2015-12-10:
I have not tested on a raspberry pi.

Are you cross compiling or compiling natively?

Honestly that does not surprise me.  Can you adjust the "build constraints" at the top of the serial_linux.go and serial_posix.go so that the posix/cgo/termios version is built instead?  If you are not familiar with build constraints, you can read more here:
https://golang.org/pkg/go/build/#hdr-Build_Constraints

Alternatively, you could try the https://godoc.org/golang.org/x/sys/unix package instead.  That package is newer than tarm/serial and I have not needed it.  Patches are welcome though.

As Rob Pike says here https://go-proverbs.github.io/
   Syscall must always be guarded with build tags.
   Cgo must always be guarded with build tags.

There are already build tags, but only by OS and not by ARCH, so probably some additions are needed for ARM and Raspberry Pi.  Patches welcome!

ngtrimble, 2015-12-10:
I have a project that uses tarm serial on Pidora on a Pi 1 B+. I compiled it on the device however. I've not had any issues with it.

fixermark, 2015-12-10:
The root issue was that I used the Go package distributed with Raspbian instead of go1.5 The Raspbian package is go 1.0.2---positively ancient.

Building 1.5 from scratch and then re-running the above "go get" command, everything worked swimmingly.

closed 2015-12-10 by fixermark