tarm/serial #36: List serial ports

cmillauriaux, 2015-12-09:
Hello,
I use your serial communication library and it works great ! Do you plan to develop a function to list available serial ports ?
Thanks in advance and well done for your work

tarm, 2015-12-10:
I have not had that need myself, but would be open to patches if you can make one.

cormacc, 2015-12-10:
I've done this on my branch -- will tidy up a bit (I'm a few commits behind master) and raise a pull request if you want?

cmillauriaux, 2015-12-10:
It will be perfect, can you make a pull request ?

cormacc, 2015-12-10:
Grand - it could be early next week before I get a chance. In the middle of some other work here I need to finish first. My updates were just to 'serial_posix' (tested on OSX and linux as far as I remember) and 'serial_windows'. Should update 'serial_linux' too, on the basis of tarm's response to another comment I made earlier -- I'd ignored it as 'serial_posix' was used by default on my linux system. Actually think the logic should be the same (just text filtering /dev/), so probably as simple as moving the method out of serial_posix into a new source file that compiles on osx or linux with or without cgo. 'serial_x' maybe?

I'll post again to this issue once I've raised the pull request

cormacc, 2015-12-14:
Raised a pull request there. GetPortsList methods are just nicked from gobug.st/serial with some minor changes -- included reference and license terms in source. But tarm may not want to merge pull request as a result. In that case you can probably just clone my fork cmillauriaux? Or use gobug.st/serial? That's a nice implementation too -- I started out using it, but switched over to tarm's library due to some stability issue (it's long enough ago that I can't remember exactly what the issue was or what platform I experienced it on :))

tarm, 2015-12-15:
Thanks @cormacc  I made some comments on your PR.  I'm going to repeat one of them here:

@cmillauriaux Can you describe the use case you have for GetPortsList()? I have never had a use for a list of all of the serial ports on the system. I suppose if you were making a GUI that required a serial port, you could present the user with a list of possible serial ports?

wolfgangmeyers, 2016-01-07:
I have a use case - I'd like to have a go application that communicates over serial port with an Arduino board. When connected it doesn't always use the same serial port (or if it does, I don't have it memorized), so it would be nice to be able to auto-discover which port the arduino board is using, even if it means attempting to open each port and test it.

cmillauriaux, 2016-01-14:
My use case is : i'm writting a cross platform application wich communicate with a device. So, in Windows, my ports are "COM1", "COM2", "COM3", etc. in Linux, my ports are "/dev/usb0" etc. In short, I can't make a list of ports hard coded, because it depends of the OS. In my app, I want to scan all of ports, try to connect to each port and when I find my device, I launch the process. I can't ask to the user, because it's a batch application.

micahwalter, 2016-08-21:
I was wondering if this was ever planned to be merged. I'd really like to be able to use it!

jfesler, 2016-12-09:
My use case: Find first available modem.  Yes, those relics are still used in places.

jkvatne, 2018-05-15:
Could you please implement the PortList function. It is not much code, and I need it very much. For now I have copied the code from https://github.com/bugst/go-serial as sugested above.

webhat, 2018-05-15:
The 2 projects are license compatible. Can you create a pull-request with you changes?

lifetyper, 2018-12-11:
I need timeout from this project,and serial port list function from bugst one,but seems neither are willing to merge what they missed.