tarm/serial #39: baud rate 2M

lubko, 2016-01-15:
please consider adding more supported baud rates,
there are few more that should work: http://play.golang.org/p/OvvdWmL56N
i have tried using 2M, and it works,
later i can provide a pull request.

tarm, 2016-01-31:
As I mentioned in https://github.com/tarm/serial/pull/37#discussion_r47597461 higher baud rates are likely not portable.  Testing on one OS is not enough.

I am open to adding higher baud rates on platforms that support it iff it can be done in a portable manner.  That might be build constraints or #defines in cgo code.

Timmmm, 2016-02-21:
It is possible (on Windows at least) to query the baud rates supported by a port. However, some ports return a list of "standard" rates but actually support arbitrary rates (e.g. I've used 111111 baud).

I don't think you can do anything other than allowing the user to set any baud rate, and failing if it doesn't succeed.