tarm/serial #40: Problems Cross Compiling [Mac + BSD]

tsteinholz, 2016-01-29:
So I am using a build tool called [gox](https://github.com/mitchellh/gox) which makes a very effortless build toolchain for all platforms. I was just wondering if serial would have support for the following platforms? They seems like they should be supported based on the code but aren't compiling. Is this something that can be fixed in the code, or does this need configuration gox side? The main build that caught me off guard was Darwin but the others would be cool too. This is the log....

```
tsteinholz@Ghost ~> gox github.com/tsteinholz/EFM32_bootloader 

Number of parallel builds: 1

-->      netbsd/arm: github.com/tsteinholz/EFM32_bootloader
-->   freebsd/amd64: github.com/tsteinholz/EFM32_bootloader
-->      darwin/386: github.com/tsteinholz/EFM32_bootloader
-->    darwin/amd64: github.com/tsteinholz/EFM32_bootloader
-->       linux/386: github.com/tsteinholz/EFM32_bootloader
-->     linux/amd64: github.com/tsteinholz/EFM32_bootloader
-->       linux/arm: github.com/tsteinholz/EFM32_bootloader
-->     freebsd/386: github.com/tsteinholz/EFM32_bootloader
-->   windows/amd64: github.com/tsteinholz/EFM32_bootloader
-->     openbsd/386: github.com/tsteinholz/EFM32_bootloader
-->   openbsd/amd64: github.com/tsteinholz/EFM32_bootloader
-->     windows/386: github.com/tsteinholz/EFM32_bootloader
-->      netbsd/386: github.com/tsteinholz/EFM32_bootloader
-->     freebsd/arm: github.com/tsteinholz/EFM32_bootloader
-->    netbsd/amd64: github.com/tsteinholz/EFM32_bootloader

10 errors occurred:
--> netbsd/arm error: exit status 2
Stderr: # github.com/tarm/serial
Development/Go/src/github.com/tarm/serial/serial.go:94: undefined: Port
Development/Go/src/github.com/tarm/serial/serial.go:95: undefined: openPort

--> freebsd/amd64 error: exit status 2
Stderr: # github.com/tarm/serial
Development/Go/src/github.com/tarm/serial/serial.go:94: undefined: Port
Development/Go/src/github.com/tarm/serial/serial.go:95: undefined: openPort

--> darwin/386 error: exit status 2
Stderr: # github.com/tarm/serial
Development/Go/src/github.com/tarm/serial/serial.go:94: undefined: Port
Development/Go/src/github.com/tarm/serial/serial.go:95: undefined: openPort

--> darwin/amd64 error: exit status 2
Stderr: # github.com/tarm/serial
Development/Go/src/github.com/tarm/serial/serial.go:94: undefined: Port
Development/Go/src/github.com/tarm/serial/serial.go:95: undefined: openPort

--> freebsd/386 error: exit status 2
Stderr: # github.com/tarm/serial
Development/Go/src/github.com/tarm/serial/serial.go:94: undefined: Port
Development/Go/src/github.com/tarm/serial/serial.go:95: undefined: openPort

--> openbsd/386 error: exit status 2
Stderr: # github.com/tarm/serial
Development/Go/src/github.com/tarm/serial/serial.go:94: undefined: Port
Development/Go/src/github.com/tarm/serial/serial.go:95: undefined: openPort

--> openbsd/amd64 error: exit status 2
Stderr: # github.com/tarm/serial
Development/Go/src/github.com/tarm/serial/serial.go:94: undefined: Port
Development/Go/src/github.com/tarm/serial/serial.go:95: undefined: openPort

--> netbsd/386 error: exit status 2
Stderr: # github.com/tarm/serial
Development/Go/src/github.com/tarm/serial/serial.go:94: undefined: Port
Development/Go/src/github.com/tarm/serial/serial.go:95: undefined: openPort

--> freebsd/arm error: exit status 2
Stderr: # github.com/tarm/serial
Development/Go/src/github.com/tarm/serial/serial.go:94: undefined: Port
Development/Go/src/github.com/tarm/serial/serial.go:95: undefined: openPort

--> netbsd/amd64 error: exit status 2
Stderr: # github.com/tarm/serial
Development/Go/src/github.com/tarm/serial/serial.go:94: undefined: Port
Development/Go/src/github.com/tarm/serial/serial.go:95: undefined: openPort
```

tarm, 2016-01-31:
Hi @tsteinholz 

This package does aim to be cross platform, but cross-compiling (with gox or manually) is tricky right now.  I think native compilation should work on all of the *bsd platforms.  For windows in particular, both native and cross compiling should work like this:

```
$ GOOS=windows go install github.com/tarm/serial
$ ls -l $GOPATH/pkg/windows_amd64/github.com/tarm
total 72
-rw-r--r--. 1 tarm tarm 72410 Jan 30 20:22 serial.a
```

Specifically the issue with cross-compiling for posix platforms is that tarm/serial uses cgo+termios.h to keep things portable.  The termios.h interface abstracts some of the syscall and ioctl differences in serial handling between different platforms.  Take a look at the "build constraints" and cgo code here:
https://github.com/tarm/serial/blob/master/serial_posix.go#L1-L9

You might be able to make it work by passing the -cgo option to gox, but cross-compiling cgo code may be tricky and I have not tried it myself.

I would be interested/open to syscall+ioctl based solutions.  Those would likely allow cross compiling more easily, but also would be much more closely tied to each architecture/platform.

Patches welcome!

dantheman213, 2020-01-01:
Hi, I have enjoyed using your serial library so far. But unfortunately I'm having this exact same issue on my own project. It's most unfortunate I can't easily create a pipeline to compile for all platforms using a single make command.

```
dan@DAN-DESKTOP:/mnt/c/Users/danie/repos/gps-atlas$ make build-macos
export GOOS=darwin && \
export GOARCH=amd64 && \
export GPSATLAS_BIN_PATH=bin/macos/gps-atlas && \
make build
make[1]: Entering directory '/mnt/c/Users/danie/repos/gps-atlas'
CGO_ENABLED=0 \
GO111MODULE=on \
go build \
-installsuffix "static" \
-ldflags="-X 'main.Version=$(cat version)'" \
-o bin/macos/gps-atlas \
$(find cmd/app/*.go)
# github.com/tarm/serial
/home/dan/go/pkg/mod/github.com/tarm/serial@v0.0.0-20180830185346-98f6abe2eb07/serial.go:128:28: undefined: Port
/home/dan/go/pkg/mod/github.com/tarm/serial@v0.0.0-20180830185346-98f6abe2eb07/serial.go:139:9: undefined: openPort
Makefile:8: recipe for target 'build' failed
make[1]: *** [build] Error 2
make[1]: Leaving directory '/mnt/c/Users/danie/repos/gps-atlas'
Makefile:29: recipe for target 'build-macos' failed
make: *** [build-macos] Error 2
dan@DAN-DESKTOP:/mnt/c/Users/danie/repos/gps-atlas$
```