tarm/serial #42: could not be installed on PCDuino

ardypro, 2016-04-05:
I attempted to install go serial on PCDuino, a Raspberry Pi like device, it threw out an error as :

_/usr/lib/go/src/pkg/github.com/tarm/serial/serial_linux.go:150: undefined: syscall.TCIOFLUSH_

evantill, 2016-04-06:
could you show us the output of `go env` on your PCDuino ?

ardypro, 2016-04-07:
GOROOT="/usr/lib/go"
GOBIN=""
GOARCH="arm"
GOCHAR="5"
GOOS="linux"
GOEXE=""
GOHOSTARCH="arm"
GOHOSTOS="linux"
GOTOOLDIR="/usr/lib/go/pkg/tool/linux_arm"
GOGCCFLAGS="-g -O2 -fPIC"
CGO_ENABLED="0"

tarm, 2016-04-08:
That constant should be present for arm after this commit:
https://github.com/golang/go/commit/bd9cd6e30e629f8751c978a75befc087b4cb5617

I'm not sure why that is not working.  I do not have an arm board to test with, but cross compiling using "GOARCH=arm" does work for me.

tarm@linux ~/go/src/github.com/tarm/serial (master)$ GOARCH=arm go install .
tarm@linux ~/go/src/github.com/tarm/serial (master)$ GOARCH=arm go env
GOARCH="arm"
GOBIN=""
GOEXE=""
GOHOSTARCH="amd64"
GOHOSTOS="linux"
GOOS="linux"
GOPATH="/home/tarm/go"
GORACE=""
GOROOT="/home/tarm/gosrc"
GOTOOLDIR="/home/tarm/gosrc/pkg/tool/linux_amd64"
GO15VENDOREXPERIMENT="1"
CC="gcc"
GOGCCFLAGS="-fPIC -marm -fmessage-length=0"
CXX="g++"
CGO_ENABLED="0"

tarm, 2016-04-08:
@ardypro Is it easy for you to reinstall packages in your GOROOT?

Specifically, I tried doing "rm -rf $GOROOT/pkg/linux_arm/*" and then "GOARCH=arm go install -x github.com/tarm/serial", and I see (with other things) this line:
mkdir -p $WORK/syscall/_obj/
cd /home/tarm/gosrc/src/syscall
/home/tarm/gosrc/pkg/tool/linux_amd64/compile -o $WORK/syscall.a -trimpath $WORK -p syscall -buildid 427ae664642404aae49a95ef68735301b88e0ca3 -D _/home/tarm/gosrc/src/syscall -I $WORK -pack -asmhdr $WORK/syscall/_obj/go_asm.h ./env_unix.go ./exec_linux.go ./exec_unix.go ./flock.go ./flock_linux_32bit.go ./lsf_linux.go ./msan0.go ./netlink_linux.go ./sockcmsg_linux.go ./sockcmsg_unix.go ./str.go ./syscall.go ./syscall_linux.go ./syscall_linux_arm.go ./syscall_unix.go ./zerrors_linux_arm.go ./zsyscall_linux_arm.go ./zsysnum_linux_arm.go ./ztypes_linux_arm.go

So that is grabbing the correct zerrors_linux_arm.go that I expect it too.  I am wondering why your compile is not finding it.  It should be there in every go version after go1.2.

evantill, 2016-04-12:
@ardypro can you check your `go version` ?

stanier, 2017-09-13:
Just to provide some resolve for other users trying to get this library to work on ARM devices-- yes it works on the RPi.

Tested with Go 1.7 on an RPi 2 cross-compiled with GOARM 5