tarm/serial #43: when read 0 bytes not return EOF on windows

xjiaj8, 2016-04-17:
c := &serial.Config{Name: "COM4", Baud: 115200, ReadTimeout: time.Second * 5}
buf:= bytes.NewBufferString("")

n, err := c.buf.ReadFrom(c.rw)
fmt.Printf("% x,%d,%v\n",buf.Bytes(),n,err)
c.buf.Reset()

tarm, 2016-04-22:
Unfortunately I don't have a windows computer to test with right now.  What do you see when you run that code?  What do you see if you do

buf := make([]byte, 1<<10)
n, err := conn.Read(buf)
fmt.Println("% x,%d,%v\n",buf[:n],n,err)

I'm not sure that the package has documented and committed to a particular return code (EOF or something with a .Timeout() method) after a timeout.

The recommended way is to just use a plain []byte slice and call c.Read() directly.

ohac, 2017-09-14:
conn.Read(buf) returns 0 and nil when timeout.

io.Reader:

> Implementations of Read are discouraged from returning a zero byte count with a nil error, except when len(p) == 0.

bufio.ReadFrom implementation is:

https://golang.org/src/bufio/bufio.go?s=16790:16849#L672

I think a zero byte with a nil error causes infinite loop.

typecampo, 2018-03-16:
time.Millisecond works for me on Windows 8.1

ReadTimeout: time.Millisecond * 5000