tarm/serial #45: windows FlowControl

gertcuykens, 2016-05-12:
https://github.com/tarm/serial/blob/master/serial.go#L111-L115

Can you add more information about 

        // RTSFlowControl bool
	// DTRFlowControl bool
	// XONFlowControl bool
	// CRLFTranslate bool

Thanks

tarm, 2016-05-12:
Those lines are commented out because they are not implemented.

A lot of serial port hardware can operate with those, but no one has added functionality to the library to be able to use that capability.  If it gets added, then it should be a complete cross-platform implementation.

One reason it has not been added is because that functionality is rarely used by the other end of the devices these days.  If you want to talk to an arduino or a robot or a modbus device or a development board, then you probably will not need or use that functionality.  It would be most useful for something like an RS232 modem for example.

Do you have a specific need or use for one of those features?

gertcuykens, 2016-05-12:
Yes for a serial rs232 device of a bridge thath give the weight of a truck
:) I will give more details when I am at my computer. Maybe on the wiki
pages there can be made more specific details and examples how to implement
flow control if needed, as far as I understand it just extra parameters for
Windows? So adding a few extra lines would cover Windows.

gertcuykens, 2016-05-12:
This are the working serial putty settings on windows
```
COM1
speed: 1200
data bits: 7
stop bits: 1
parity: even
flow control: DSR/DTR
```

gertcuykens, 2016-05-12:
never mind seem to be working regardless the DSR/DTR stuff

closed 2016-05-12 by gertcuykens