tarm/serial #46: Windows compatibility

phoenix147, 2016-05-30:
Hi!

I have a strange problem, which gives me serious headaches. I am working with an device, which is connected through an FTDI chip. It has some quirks itself (for example only answering aber 20 seconds after accessing the serial port), but i managed to build a go library for it and it works pretty well, on linux.

As soon as i want to use the library on windows, i won't get any answer from the probe. I've tested the serial port itself with python, c and putty and it works flawlessly. It requires no special configuration, just 19200 baud, 8 data bits, 1 stop bit, no parity, no hardware flow control. I've even implemented a solution with cgo, which also works flawlessly.

I also tried every other golang serial library i've found, and all of them have the same problem. It's also interesting, that the serial library in general is working with other devices on windows.

Can you maybe give me a hint in the right direction? I'm sorry i can't give any more detail information on this, but i'm not quit sure what could be useful.

thanks in advance

larsgk, 2016-08-30:
"Only answering after 20 secs" sounds like modemmanager tries to grab it.  Add it to the udev usb blacklist (VID/PID)

phoenix147, 2016-08-30:
the problem with the 20 secs is related to the device itself, because it has some blocking timeouts in its startup routine and has nothing to do with the serial connection

tarm, 2016-08-30:
Is there something like "strace" for windows?  It would be interesting to compare the go version with a program that works.

Maybe something like https://msdn.microsoft.com/en-us/library/windows/hardware/ff552060(v=vs.85).aspx or https://technet.microsoft.com/en-us/sysinternals/bb896645.aspx

Sorry, I don't use windows much (and don't have a windows computer or VM to test with at the moment).

Timmmm, 2016-09-05:
Do other serial programs on Windows work (e.g. Termite)?

KickinWingAnimalDoctor, 2018-08-24:
Experiencing this problem with Windows as well.

phoenix147, 2019-02-14:
It seems that the device i was using requires some sort of Flow control. I found out by trying termite on Windows, which also only worked when using RTS/DTS. It seems that other libraries and programs, e.g. putty, are setting this implicitly.

Using another library (https://github.com/jacobsa/go-serial) with support for Flow control seems to be working better. It would be nice to have support for flow control in this library too, see #96