tarm/serial #59: mipsle compile

dfense, 2017-05-28:
Hi,

Your library looks solid and can use on linux np. However, I did most recently purchase some new IoT Omega2 processors from onion.io and they could be quite popular. 

I have been writing programs in GO and running on it w/no problem. Runs great, small and tight.
I was hoping to get some serial port development going on it, and of course, seem to be quite a few platform specific items breaking. 

Your serial lib seems the closest. I try to compile the simple default sample you provide with the library, and get the following error using this compile syntax.

 GOOS=linux GOARCH=mipsle go build -compiler gc serial.go

```
# github.com/tarm/serial
src/github.com/tarm/serial/serial_linux.go:105: unknown field 'Ispeed' in struct literal of type syscall.Termios
src/github.com/tarm/serial/serial_linux.go:106: unknown field 'Ospeed' in struct literal of type syscall.Termios
```

Anything I can do to help you support this platform ?

tarm, 2017-05-29:
Looks the same as this issue:
https://github.com/golang/go/issues/18866

Looks like there are some suggested approaches in that thread.

I would be open to patches that specialize with build constraints based on GOARCH (we have build constraints based on GOOS and CGO right now, but not GOARCH).

ljubomirb, 2017-11-27:
the same issue with GOARCH=mips
(using Go 1.9.2)

closed 2017-12-15 by tarm

tarm, 2018-01-14:
@nocko I believe this is compiling as expected on mips.  I just pushed a change to the travis.yml file that adds cross compile targets for arm, mips, and mipsle and those at least compile (I have not personally tested mips, but arm and x64 are in production).

commit fdb95b50 (2017-12-15): Merge pull request #73 from hnw/switch-to-x_sys_unix

Use "golang.org/x/sys/unix" instead of "syscall" (fix #59)