tarm/serial #63: cannot read a whole message

joyqat, 2017-08-21:
```go
buf := make([]byte, 128)
for {
	n, err = s.Read(buf)
	if err != nil {
	    log.Fatal(err)
	}
	log.Printf("%q", buf[:n])
}
```
result of code above:
when receive "abcdef"
it will print 
"a" and "bcdef"
always a single byte followed by reminded bytes.
what is the problem?

btw, I use windows 10 build1703, Go 1.8, VSPD 6.9(virtual serial port).

Robin3D, 2017-08-24:
Me too, I have the same problem

paocalvi, 2017-08-24:
My two cents.....
IMHO, this approach to the reading is not correct. Like in TCP, you should
not relay on a bunch of bytes arriving all together just because they have
been sent together, because of fragmentation, when it comes to serial, you
should be prepared to received half (or part) of a message, and then the
rest, or many messages together. Even if in the case of a directly
connected serial-to-serial your expectations is correct, it is certainly
not when you for exemple, pass through some serial-over-ip device or SW. If
you try in any case to write SW not relying on "all message delivered at
once" the result will be more robust and reusable.

Paolo

joyqat, 2017-08-24:
thanks for your opinion

closed 2017-08-24 by joyqat