tarm/serial #71: Timeouts

pascaldekloe, 2017-11-01:
The API documentation notes *"There is currently no exposed way to set the timeouts, though patches are welcome."*. On the other hand response on issues and pull requests is minimal to say the least. So @tarm, are you still maintaining this project?

## Proposal

How about the following 3 methods from `net.Conn`?

```
    // SetDeadline sets the read and write deadlines associated
    // with the connection. It is equivalent to calling both
    // SetReadDeadline and SetWriteDeadline.
    //
    // A deadline is an absolute time after which I/O operations
    // fail with a timeout (see type Error) instead of
    // blocking. The deadline applies to all future and pending
    // I/O, not just the immediately following call to Read or
    // Write. After a deadline has been exceeded, the connection
    // can be refreshed by setting a deadline in the future.
    //
    // An idle timeout can be implemented by repeatedly extending
    // the deadline after successful Read or Write calls.
    //
    // A zero value for t means I/O operations will not time out.
    SetDeadline(t time.Time) error

    // SetReadDeadline sets the deadline for future Read calls
    // and any currently-blocked Read call.
    // A zero value for t means Read will not time out.
    SetReadDeadline(t time.Time) error

    // SetWriteDeadline sets the deadline for future Write calls
    // and any currently-blocked Write call.
    // Even if write times out, it may return n > 0, indicating that
    // some of the data was successfully written.
    // A zero value for t means Write will not time out.
    SetWriteDeadline(t time.Time) error
```

pascaldekloe, 2017-11-09:
Unmaintained indeed. For now I made timeouts work with https://github.com/pascaldekloe/nbio.

tarm, 2017-11-09:
The package works for me and is feature complete for my needs, so I am not actively making changes to it.  The package has and does transfer millions of bytes in production systems and I have an active interest in making sure it continues to do so.

From my perspective, it does what I need it to, but if other people have other needs, then pull requests that are well designed and work across all platforms will be accepted.

Read timeouts already exist in the package, but they are have a different API than net.Conn.  Search for timeout in https://godoc.org/github.com/tarm/serial  The documentation of timeouts could be better (part of the docs say that still need to be implemented).  The low level serial timeouts work differently than socket based timeouts and work differently on different platforms, so I suspect implementing the net.Conn timeout API would either be very complicated or would be misleading to people.

pascaldekloe, 2017-11-09:
Now I see that the read timeout is per request. 🔔

“Implementations of Read are discouraged from returning a zero byte count with a nil error”
 … and:
“Callers should treat a return of 0 and nil as indicating that nothing happened”
 — https://godoc.org/io#Reader

Therefore a dedicated error is required such as the following.
https://godoc.org/github.com/pascaldekloe/nbio#ErrNoData

Do you agree @tarm?

PS
Maybe the "works for me" and the pull request barrier should be in the README to prevent disappointment? ;-)

rugginoso, 2017-11-16:
@pascaldekloe if you wait a little (go 1.10 I guess) you will get this for free: https://go-review.googlesource.com/c/go/+/71770

pascaldekloe, 2017-11-16:
Indeed, the proposed convention seems to be the way to go. However I'm still waiting on @tarm to respond, again…

jonaz, 2023-12-05:
status on this one @tarm ?