tarm/serial #81: Windows ReadTimeout no effect

MiyamuraMiyako, 2018-04-17:
when serial readbuffer haven't any data. ```Read``` return immediately, not wait setting's timeout.

MiyamuraMiyako, 2018-04-17:
Just simple code. @xiegeo 

```go
sc := &serial.Config{Name: "COM3", Baud: 9600, Size: 8, Parity: serial.ParityNone, StopBits: serial.Stop1, ReadTimeout: time.Millisecond * 50}
sp, err := serial.OpenPort(sc)
if err != nil {
	fmt.Println(err.Error())
	return false
}
buf := make([]byte, 128)
sp.Read(buf)
return true
```

xiegeo, 2018-04-17:
50 milliseconds is very short, so it's unlikely to catch anything before timing out.

Also, your read doesn't do anything, you should check for errors and the number returned to know if you have read any data. 

As your code is written,  your behavior is expected.

MiyamuraMiyako, 2018-04-17:
@xiegeo sorry,My mistake.

closed 2018-04-17 by MiyamuraMiyako