tarm/serial #84: Serial  only receive 1 byte

andarm, 2018-07-26:
### run on window  
-  used modusslave-tool test . 
``` 
	sendbuf := []byte{0x40, 0x02, 0x00, 0x00, 0x00, 0x6E, 0xF6, 0xF7}
		n, err := s.Write(sendbuf)
	if err != nil {
		s.Close()
		fmt.Println("write close")
		log.Fatal(err)
	}

	buf := make([]byte, 128)

	n, err = s.Read(buf)
	if err != nil {
		fmt.Println("read close")
		log.Fatal(err)
	}
``` 
   - receive len ｎ＝１　byte .   // Normal receive is : 40 02 0E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 4E C1 ,but I only got one byte. 
   - serial config  
``` 
  	c := &serial.Config{Name: "COM5", Baud: 9600, ReadTimeout: time.Second * 5}
``` 
###

xiegeo, 2018-07-27:
You need to put read in a loop. Use the third byte to calculate total length.

If you need something that also takes timing into consideration and handles more opcodes, you can check out my code here https://github.com/xiegeo/modbusone/blob/master/packet_reader.go

andarm, 2018-07-28:
Thanks ,I got it.