tarm/serial #88: Having 0x10 set causes arduino to reset when serial message is sent

ASoggyToaster, 2018-09-02:
```
func setCommState(h syscall.Handle, baud int, databits byte, parity Parity, stopbits StopBits) error {
	var params structDCB
	params.DCBlength = uint32(unsafe.Sizeof(params))

	params.flags[0] = 0x01  // fBinary
	params.flags[0] |= 0x10 // Assert DSR
```

Having 0x10 set causes arduino to reset when serial message is sent when connecting via windows. I modified my copy of the library to remove this line.

```
func setCommState(h syscall.Handle, baud int, databits byte, parity Parity, stopbits StopBits) error {
	var params structDCB
	params.DCBlength = uint32(unsafe.Sizeof(params))

	params.flags[0] = 0x01  // fBinary
	//params.flags[0] |= 0x10 // Assert DSR
```

Currently using it in this project: https://github.com/ASoggyToaster/Max7219

ASoggyToaster, 2018-09-02:
Submitted a PR: https://github.com/tarm/serial/pull/89

donotnoot, 2019-02-20:
jeez I wasted so much time yesterday trying to figure out why my Arduino was resetting when I sent data using this library. Thanks for finding out wha the problem is.

ASoggyToaster, 2019-03-28:
Agreed - Probably shouldn't have submitted this as a PR per se, but having it open - I think - is helpful to others who may experience the same issue. 

In this instance I made the quick and dirty change that suited my usecase, but glad it helped you as well @donotnoot !