tarm/serial #94: 250K baud

charles-d-burton, 2019-01-01:
I'm trying to adapt this code to support a 250000 baud rate for 3D printers.  Unfortunately I don't fully understand how I would interface with the kernel to achieve this.  Mostly I'm trying to use idiomatic go but the syscall package doesn't support non-standard baud rates.  My plan was to build a function that will handle non-standard baud rates then rather than throwing an error and would instead emit a warning about using a non-standard baud rate, then attempt to set it.  Is there anything someone can point me to that would allow for this?

xiegeo, 2019-01-05:
Which operating system are you using?  You should just modify the serial_[os].go file, otherwise, you are just reimplementing the entire feature.

charles-d-burton, 2019-01-07:
I'm using raspbian, the problem is that this project uses the built in serial interface for Linux which only supports standard baud rates.  There's a newer interface called termios2 in the kernel that allows arbitrary baud rates.  I found a project that does work with termios2 from golang and I've tested it against my 3D printer, if anyone is interested I can port the work over here and make a PR so that this project can support more arbitrary baud rates in Linux.

mpictor, 2019-03-22:
Have you actually tried 250k? It's in the source, so seems safe to assume it is supported - see line 42 of
https://github.com/tarm/serial/blob/master/serial_linux.go#L42

kris14an, 2019-12-11:
@charles-d-burton did you try port this library to terminos2 to support 250k baudrate?

charles-d-burton, 2019-12-11:
> @charles-d-burton did you try port this library to terminos2 to support 250k baudrate?

I did, the code is over here:
https://github.com/charles-d-burton/serinit

It supports auto-detection or you can specify the device with the struct that I export.  Always open to improvements and PRs if you're interested.