tarm/serial #101: windows10 s.Read(buffer) is empty

hinet, 2019-05-23:
`
config := &serial.Config{Name: "COM3", Baud: 2400, Parity: 'E', StopBits: 1, ReadTimeout: time.Second * 5}
port, err := serial.OpenPort(config)
checkErr(err)
defer port.Close()
n, err := port.Write([]byte("FEFEFEFE68310010111819681104333334333516"))
checkErr(err)

//read
n, err = port.Read(buffer)
checkErr(err)
log.Printf("read：%q", buffer[:n])
`

hinet, 2019-05-24:
```
//I added sleep. It's OK.
time.Sleep(1 * time.Second)
buffer := make([]byte, 1024)
n, err = port.Read(buffer)
checkErr(err)
log.Printf("read：%q", buffer[:n])
```
I added sleep. It's OK.Is there any other way?

xiegeo, 2019-05-24:
Read in a loop until you got everything. If read repeatedly returns early, add a sleep calculated from expected data length and baud rate.