tarm/serial #105: Can't compile on MacOS

dantheman213, 2020-01-01:
Hi, I recently commented in https://github.com/tarm/serial/issues/40. I see the issue isn't just cross compiling... I'm on a Macbook Pro right now and can't compile natively either. The README doesn't specify it can't compile on MacOS, quite the opposite it implies that it can be compiled on any modern desktop or server OS.

```
gps-atlas dan$ CGO_ENABLED=0 \
> GO111MODULE=on \
> go build \
> -installsuffix "static" \
> -ldflags="-X 'main.Version=$(cat version)'" \
> -o bin/macos/gps-atlas \
> $(find cmd/app/*.go)
# github.com/tarm/serial
../../../go/pkg/mod/github.com/tarm/serial@v0.0.0-20180830185346-98f6abe2eb07/serial.go:128:28: undefined: Port
../../../go/pkg/mod/github.com/tarm/serial@v0.0.0-20180830185346-98f6abe2eb07/serial.go:139:9: undefined: openPort
```

xiegeo, 2020-01-02:
try `CGO_ENABLED=1`

dantheman213, 2020-01-03:
@xiegeo thank you that worked.

closed 2020-01-03 by dantheman213

liwenbo, 2023-11-24:
NB