tarm/serial #127: all serials

gitchander, 2023-01-19:
Hello! How I can get all serial ports?

jd-boyd, 2023-03-24:
It would be extremely helpful if a method to list available serial ports was provided.

FObersteiner, 2023-04-20:
something [like this](https://stackoverflow.com/a/14224477/10197418) might be easy to translate to `go`, however I'm not sure how reliable this is. Does it show ports that are unused (some machines still have dedicated COM ports)? Are all ttys globed like this actually serial ports?

oneAsiaPeople, 2023-09-07:
> 

Thank you very much for the solution you provided, but when the program is compiled into an executable file, the execution platform may not have GOOS environment variables. Does this also need to be considered