tarm/serial #128: There is no error ReadTimeout

gitchander, 2023-02-20:
`package main

import (
	"fmt"
	"log"
	"time"

	"github.com/tarm/serial"
)

func checkError(err error) {
	if err != nil {
		log.Fatal(err)
	}
}

func main() {

	config := &serial.Config{
		//Name:        "COM1",
		Name:        "/dev/ttyUSB0",
		Baud:        115200,
		ReadTimeout: 2 * time.Second,
	}
	p, err := serial.OpenPort(config)
	checkError(err)
	defer p.Close()

	data := []byte("ping")
	_, err = p.Write(data)
	checkError(err)

	res := make([]byte, 1024)

	n, err := p.Read(res) // ???
	checkError(err)

	fmt.Printf("readed %d bytes, hex: [%x]\n", n, res[:n])
}
`
If serial port not return bytes:
for Linux it returns EOF;
for windows it returns readed 0 bytes error is nil.
If I used bufio reader ReadByte() - operation is blocking because serial Read retuns -> (0 , nil).

FObersteiner, 2023-03-07:
I think I have experienced this the other way around, when running code on Linux that I had developed on Windows. Suddenly, I had to catch an error. In general, I agree that it should be a ReadTimeout error, on all platforms to have consistent behavior.

FObersteiner, 2023-03-07:
I had the same problem lately; unfortunately this repo here does not seem to be actively maintained.

However, I think some of the forks implement useful extensions. For example https://github.com/urbanminded/serial seems to have what we're looking for in [59f58d4f6eb90c61c481fff26c5040f2bf2a43d7](https://github.com/tarm/serial/commit/59f58d4f6eb90c61c481fff26c5040f2bf2a43d7)

gitchander, 2023-03-13:
Thanks for comment!

closed 2023-03-13 by gitchander