tarm/serial #129: not clear buffer splitting when reading

thedemoncat, 2023-03-18:
HI!!!

arduino App (platformio - esp8266)

```
#include <Arduino.h>
#include <ESP8266WiFi.h>

void setup() {
  // initialize serial:
  Serial.begin(9600);

}

void loop() {
  Serial.print("Hello world");
  delay(10000);   
}
```

golang app

```
package main

import "fmt"
import "log"
import "github.com/tarm/serial"

func main() {
	config := &serial.Config{
		Name: "COM3",
		Baud: 9600,
	}

	stream, err := serial.OpenPort(config)
	if err != nil {
		log.Fatal(err)
	}

	buf := make([]byte, 1024)

	for {
		n, err := stream.Read(buf)
		if err != nil {
			log.Fatal(err)
		}
		s := string(buf[:n])
		if s == "" {
			continue
		}
		fmt.Println(s)
	}
}

```

in the console I get this result:

```
API server listening at: 127.0.0.1:57847
H
ello world
H
ello world

```

Why is the first character read separately in the stream?

How to read a line correctly?

Thank you in advance!

orestonce, 2023-03-27:
bufio.NewReader(stream).ReadLine()

slinky55, 2024-04-20:
Is there a way to set a custom timeout on bufio.NewReader()? I've set a timeout on the port itself as well

FObersteiner, 2024-04-21:
> Is there a way to set a custom timeout on bufio.NewReader()? I've set a timeout on the port itself as well

the underlying io.Reader does not implement a timeout - the idea here being that the reader finishes when the resource is closed or exhausted (EOF). So the correct way I think is to set the timeout on the port, which is your resource in this case. Note however that on Windows, this does not work with the original tarm/serial. There are a couple of forks that implement this feature ([example](https://github.com/tarm/serial/commit/1b85a7461b184c65a5d81fb7e69897820b9b0b31)).