tarm/serial #132: 为什串口有数据时，读取的数据总是将一帧发送数据的第一字节和剩余的字节数，分2次才能完全的读到？

ruixule, 2024-06-11:


ruixule, 2024-06-11:
[image]
[image]
c := &serial.Config{
		Name:        "COM3",
		Baud:        9600,
		Parity:      0,
		StopBits:    1,
		ReadTimeout: 100 * time.Millisecond,
	}
	port, err := serial.OpenPort(c)

	if err != nil {
		u.Lg.Error("Failed to open port:", err)
		return
	}
	defer port.Close()

	var receivedData []byte

	for {
		data := make([]byte, bufferSize) // 创建一个固定大小的缓冲区
		n, err := port.Read(data)
		if err != nil {
			u.Lg.Error(err)
			errCount++
			if errCount > maxErrCount {
				u.Lg.Error("errCount > maxErrCount")
				return
			}
		} else {
			errCount = 0
			if n > 0 {
				//receivedData = append(receivedData, data[:n]...)
				u.Lg.Infof("Received data so far: %v", data[:n])
				if n > processBytes {
					processSerialData(receivedData)
					receivedData = nil // 重置接收的数据
				}
			}
		}
	}

tocurd, 2025-01-14:
这个正常，你需要自己要做好拼接工作

FieldsYeh, 2025-04-19:
increase timeout to 1 second